STMicroelectronics/OpenOCD #1: STM32F769 based board with different QSPI pins

ladislas, 2020-04-16:
Hi there --

We are working on a custom board using an STM32F769BI and might choose different pins for the QSPI flash.

I'm wondering how we would need to change [`proc qspi_init { }`](https://github.com/STMicroelectronics/OpenOCD/blob/openocd-cubeide-r1/tcl/board/stm32f769i-disco.cfg#L18) for it work on different pins?

Do we need to change other things as well?

Thanks for your help!

tarek-bochkati, 2020-04-17:
Hi Ladislas,

Using [stm32_gpio_conf.pl](https://github.com/STMicroelectronics/OpenOCD/blob/openocd-cubeide-r1/contrib/stm32_gpio_conf.pl) you can generate the needed code to put into qspi_init.

```
# Helper for generating GPIO setup for STM32F0, F4, F7, H7, L0, L1, L4, L4+
# and F1 (for 'stmqspi' and 'msoftspi' drivers).
#
# Each pin is configured by "PortAndBit:Conf:Speed"
#  'PortAndBit' specifies Port and bit number
#  'Conf' is one of 'AFx' (alternate), 'P' (output), 'IN' (input),
#    (each optionally by 'P' (push-pull) or 'O' (open-drain)),
#    (all optionally followed by 'UP' (pull-up), or 'DO' (pull-down))
#  'Speed' is one of 'L' (low), 'M' (medium), 'H' (high), 'V' (very high)
#
# Port configuration can be given on command line as a single string (pins separated by commas)
# or via CubeMX generated file. The latter must consist of the quadspi.c / octospi.c and the
# corresponding header. The precise spelling in these files doesn't seem to be consistent, though ...
#
```

in the same file you can find some examples:
```
# stm32f769i-disco
#$GPIO_BASE = 0x40020000;
#$Conf = "PB06:AF10:V, PB02:AF09:V, PC10:AF09:V, PC09:AF09:V, PD13:AF09:V, PE02:AF09:V";
# w/ msoftspi
#$Conf = "PB06:PP:M, PB02:PP:V, PD13:INUP:V, PE02:INUP:V, PC10:INUP:V, PC09:INUP:V, ";
```

**PS**: please consider creating a new cfg file with your change

Kind Regards,
Tarek

ladislas, 2020-04-17:
Thanks a lot @tarek-bochkati for the answer, I'll look into that! :)

closed 2020-04-21 by tarek-bochkati