STMicroelectronics/OpenOCD #2: Out of bound error

nretro, 2020-06-07:
In src/flash/mflash.h the line

`mg_io_uint8 reserved7[186];`

should be changed to

`mg_io_uint8 reserved7[190];`

This is obviously an error, since in mflash.c reserved7 is accessed using
` memset(pSegIdDrvInfo->reserved7, 0x00, 190);`

Current gcc compiler flags this with an out of bound warning. In combination with the "Werror" flag, the compilation will be aborted.

tarek-bochkati, 2020-06-16:
Thanks @nretro 

Could you please specify the GCC version you are using.
BTW, we are doing an alignment with official OpenOCD : [github mirror](https://github.com/ntfreak/openocd)
In the official repo the mflash driver is deleted : please check this [link](https://github.com/ntfreak/openocd/tree/master/src/flash) and the [deletion commit](https://github.com/ntfreak/openocd/commit/7a93c9e087b7cb0444068a3deb506eb127f758f0)
So I would like the have the compiler version you used, so I can check on the official OpenOCD.

Thanks again,
Tarek

nretro, 2020-06-16:
Hi Tarek,

thanks for your answer. I'm using "gcc (Debian 9.3.0-13) 9.3.0" on debian unstable. 
The official openocd compiles nicely and also supports both my chinese ST-LINK V2 clone and my STM32H7 Nucleo board. Maybe after more than three years, it would be time for another upstream release... the 0.10.0 release (which comes with debain) of 2017 does have problems on both boards.

Thanks,
Norbert.

tarek-bochkati, 2020-06-18:
Thanks again Norbert

Our next release of CubeIDE planned beginning of July will contain the latest official OpenOCD code plus STMicroelectronics additions (not merged in official openocd yet).
And this fork does not include the mflash driver.

> the 0.10.0 release (which comes with debain) of 2017 does have problems with both boards.

indeed, I have started last year upstreaming ST code so it should be OK, you can check this [link](http://openocd.zylin.com/#/q/owner:tarek.bouchkati%2540gmail.com+status:merged)
and we are still reviewing the commits adding the latest ST devices : [link](http://openocd.zylin.com/#/q/owner:tarek.bouchkati%2540gmail.com+status:open)
By the way, we are talking about a new release 0.11, hopefully it can be done before the end of this summer:
0.11.0-rc1 proposal : http://openocd.zylin.com/#/c/5707/

Please do not hesitate to share with us, if you detect any issue or if you feel that there is something missing.

Thanks,
Tarek

stansotn, 2020-11-24:
Just tried to build on `gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0` arm64 raspberry pi 4 and ran into the same issue. After patching the following compilation goes further. Thanks @nretro 

> In src/flash/mflash.h the line
> 
> `mg_io_uint8 reserved7[186];`
> 
> should be changed to
> 
> `mg_io_uint8 reserved7[190];`

Then breaks here
```
/usr/include/aarch64-linux-gnu/sys/sysctl.h:21:2: error: #warning "The <sys/sysctl.h> header is deprecated and will be removed." [-Werror=cpp]
   21 | #warning "The <sys/sysctl.h> header is deprecated and will be removed."
      |  ^~~~~~~
cc1: all warnings being treated as errors
```
Build succeeded only if `-Werror` is removed from the makefile.

tarek-bochkati, 2021-02-05:
this should be fixed in https://github.com/STMicroelectronics/OpenOCD/tree/openocd-cubeide-r2

closed 2021-02-05 by tarek-bochkati