STMicroelectronics/OpenOCD #21: STM32F767ZG says "Cortex-M PARTNO 0x0 is unrecognized"

rusefillc, 2023-06-25:
I am unable to connect to STM32F767ZG 

```
Open On-Chip Debugger 0.12.0-00017-gb153daa14 (2023-02-03-14:58) [https://github.com/STMicroelectronics/OpenOCD]
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : clock speed 4000 kHz
Info : STLINK V2J34S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.254183
Error: [stm32f7x.cpu] Cortex-M PARTNO 0x0 is unrecognized
Warn : target stm32f7x.cpu examination failed
Info : starting gdb server for stm32f7x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: [stm32f7x.cpu] Cortex-M PARTNO 0x0 is unrecognized
Error: Target not examined yet

```

rusefillc, 2023-06-25:
with debug
```
Open On-Chip Debugger 0.12.0-00017-gb153daa14 (2023-02-03-14:58) [https://github.com/STMicroelectronics/OpenOCD]
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 4 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 4 options.c:52 configuration_output_handler(): 
Debug: 5 4 options.c:233 add_default_dirs(): bindir=/src/work/openocd/win32/build/bin
Debug: 6 4 options.c:234 add_default_dirs(): pkgdatadir=/src/work/openocd/win32/build/share/openocd
Debug: 7 5 options.c:235 add_default_dirs(): exepath=C:/stuff/rusefi/misc/install/openocd
Debug: 8 5 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 5 configuration.c:33 add_script_search_dir(): adding C:/Users/Dell2019/AppData/Roaming/OpenOCD
Debug: 10 5 configuration.c:33 add_script_search_dir(): adding C:/stuff/rusefi/misc/install/openocd/../share/openocd/site
Debug: 11 6 configuration.c:33 add_script_search_dir(): adding C:/stuff/rusefi/misc/install/openocd/../share/openocd/scripts
Debug: 12 6 command.c:155 script_debug(): command - ocd_find openocd/st_nucleo_f7.cfg
Debug: 13 6 configuration.c:88 find_file(): found openocd/st_nucleo_f7.cfg
Debug: 14 8 command.c:155 script_debug(): command - ocd_find openocd/stlink.cfg
Debug: 15 8 configuration.c:88 find_file(): found openocd/stlink.cfg
Debug: 16 8 command.c:155 script_debug(): command - adapter driver hla
Debug: 17 9 command.c:155 script_debug(): command - hla_layout stlink
Debug: 18 9 hla_interface.c:230 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 19 9 command.c:155 script_debug(): command - hla_device_desc ST-LINK
Debug: 20 9 hla_interface.c:217 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 21 9 command.c:155 script_debug(): command - hla_vid_pid 0x0483 0x3744 0x0483 0x3748 0x0483 0x374b 0x0483 0x374d 0x0483 0x374e 0x0483 0x374f 0x0483 0x3752 0x0483 0x3753 0x0483 0x3754
Debug: 22 10 command.c:155 script_debug(): command - transport select hla_swd
Debug: 23 10 hla_transport.c:191 hl_swd_transport_select(): hl_swd_transport_select
Debug: 24 10 command.c:155 script_debug(): command - ocd_find openocd/stm32f7x.cfg
Debug: 25 11 configuration.c:88 find_file(): found openocd/stm32f7x.cfg
Debug: 26 11 command.c:155 script_debug(): command - ocd_find openocd/swj-dp.tcl
Debug: 27 11 configuration.c:88 find_file(): found openocd/swj-dp.tcl
Debug: 28 12 command.c:155 script_debug(): command - transport select
Debug: 29 12 command.c:155 script_debug(): command - ocd_find openocd/mem_helper.tcl
Debug: 30 12 configuration.c:88 find_file(): found openocd/mem_helper.tcl
Debug: 31 12 command.c:155 script_debug(): command - add_usage_text mrw address
Debug: 32 13 command.c:155 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 33 13 command.c:155 script_debug(): command - add_usage_text mrh address
Debug: 34 13 command.c:155 script_debug(): command - add_help_text mrh Returns value of halfword in memory.
Debug: 35 13 command.c:155 script_debug(): command - add_usage_text mrb address
Debug: 36 14 command.c:155 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 37 14 command.c:155 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 38 14 command.c:155 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 39 15 command.c:155 script_debug(): command - transport select
Debug: 40 15 command.c:155 script_debug(): command - expr  [ string first "jtag" $_TRANSPORT ] != -1 
Debug: 41 15 command.c:155 script_debug(): command - transport select
Debug: 42 15 command.c:155 script_debug(): command - expr  [ string first "jtag" $_TRANSPORT ] != -1 
Debug: 43 16 command.c:155 script_debug(): command - transport select
Debug: 44 16 command.c:155 script_debug(): command - expr  [ string first "swd" $_TRANSPORT ] != -1 
Debug: 45 16 command.c:155 script_debug(): command - swd newdap stm32f7x cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id 0x5ba02477
Debug: 46 17 hla_tcl.c:102 jim_hl_newtap_cmd(): Creating New Tap, Chip: stm32f7x, Tap: cpu, Dotted: stm32f7x.cpu, 8 params
Debug: 47 17 hla_tcl.c:112 jim_hl_newtap_cmd(): Processing option: -irlen
Debug: 48 17 hla_tcl.c:112 jim_hl_newtap_cmd(): Processing option: -ircapture
Debug: 49 17 hla_tcl.c:112 jim_hl_newtap_cmd(): Processing option: -irmask
Debug: 50 18 hla_tcl.c:112 jim_hl_newtap_cmd(): Processing option: -expected-id
Debug: 51 18 core.c:1478 jtag_tap_init(): Created Tap: stm32f7x.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
Debug: 52 18 command.c:155 script_debug(): command - dap create stm32f7x.dap -chain-position stm32f7x.cpu
Debug: 53 18 command.c:155 script_debug(): command - target create stm32f7x.cpu cortex_m -endian little -dap stm32f7x.dap -ap-num 0 -gdb-max-connections 2
Info : 54 19 target.c:6205 target_create(): The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Debug: 55 19 hla_target.c:193 adapter_target_create(): adapter_target_create
Debug: 56 20 hla_target.c:163 adapter_init_arch_info(): adapter_init_arch_info
Debug: 57 20 command.c:289 register_command(): command 'rtt' is already registered
Debug: 58 20 command.c:289 register_command(): command 'tpiu' is already registered
[... cut ...]
Debug: 121 69 hla_layout.c:36 hl_layout_open(): failed
Debug: 122 69 command.c:545 run_command(): Command 'transport init' failed with error code -4
User : 123 69 command.c:608 command_run_line(): 
Debug: 124 69 command.c:545 run_command(): Command 'init' failed with error code -4
User : 125 69 command.c:608 command_run_line(): 
Debug: 126 70 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 127 70 hla_interface.c:119 hl_interface_quit(): hl_interface_quit
```

Arslan8, 2024-10-21:
Any updates on this?

freebsdly, 2025-05-13:
have the same problem on stm32f429igt6 board, after downloading a zephyr demo

mrjimenez, 2025-09-22:
Same problem here on a `st_nucleo_h745zi` and `stm32h747i-disco`.

Before I have loaded the executable, openocd could recognize the Cortex-M. After loading, no longer:

```bash
$ openocd -f /usr/share/openocd/scripts/board/stm32h747i-disco.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control.
       The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1800 kHz
Info : STLINK V3J16M9 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.272439
Info : [stm32h747xih6.cpu0] Cortex-M7 r1p1 processor detected
Info : [stm32h747xih6.cpu0] target has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h747xih6.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

$ openocd -f /usr/share/openocd/scripts/board/stm32h747i-disco.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control.
       The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1800 kHz
Info : STLINK V3J16M9 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.272871
Error: [stm32h747xih6.cpu0] Cortex-M PARTNO 0x0 is unrecognized
Warn : target stm32h747xih6.cpu0 examination failed
Info : starting gdb server for stm32h747xih6.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

```

jesultra, 2026-08-17:
I am seeing `Error: [at91sam3X8E.cpu] Cortex-M PARTNO 0x0 is unrecognized` on a LPC1343, after initially being able to load an image.

I suspect that the reason could be that the image loaded could be putting the CPU into sleep or low-power mode while idle, which is generally good practice for production systems.  As far as I know, part of the JTAG/SWD handling is also powered down when the CPU is.

I have yet to find a sequence of using external RESET and SWD to be able to reset the CPU, and then get to "reset halt" via OpenOCD before the CPU has a chance to execute any instructions from its internal flash (which could e.g. disable the SWD pins, put the CPU into sleep, reconfigure the PLL to invalid settings, or by other means cause the SWD interface to stop working within microseconds of booting).