STMicroelectronics/OpenOCD #27: STM32H5 Error: The selected transport doesn't support this target

DBY29681, 2024-05-31:
Open On-Chip Debugger 0.12.0-00024-g34afaa939 (2024-05-31-15:10) [https://github.com/STMicroelectronics/OpenOCD]  
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Error: The selected transport doesn't support this target
D:/Environment/MSYS64/mingw64/bin/../share/openocd/scripts/target/stm32h5x.cfg:55: Error:
in procedure 'script'
at file "embedded:startup.tcl", line 28
at file "D:/Environment/MSYS64/mingw64/bin/../share/openocd/scripts/target/stm32h5x.cfg", line 55

MartinPoupa, 2024-08-29:
Hello, i have similar problem. I am training flesh stm32H503 via STM32CubeMX and VSCode on ubuntu 22. Thanks you for help.
 ```
Open On-Chip Debugger 0.12.0-00024-g34afaa939 (2024-08-29-13:52) [https://github.com/STMicroelectronics/OpenOCD]
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Error: The selected transport doesn't support this target
/usr/local/bin/../share/openocd/scripts/target/stm32h5x.cfg:55: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 28
at file "./openocd.cfg", line 8
at file "/usr/local/bin/../share/openocd/scripts/target/stm32h5x.cfg", line 55

cosmikwolf, 2024-09-11:
Having this same problem, with the STM32H563 Nucleo 144 board

cosmikwolf, 2024-09-12:
I found that if I set the CHIPNAME to something that wasn't stm32h5x, it diddn't like it. I had it set to STM32H563 before. I just removed the chipname setting all together, since it is set in the stm32h5x.cfg and it works now

DBY29681, 2024-09-14:
62/10000
实时翻译
你能分享你的配置吗？我记得将目标配置为stm32h5x，遇到了链接脚本在openocd目标下找不到目标的错误

划译
Can you share your configuration? I remember configuring the target as stm32h5x and encountering this error where the linking script cannot find the target under the openocd target

DBY29681, 2024-09-14:
Can you share your configuration? I remember configuring the target as stm32h5x and encountering this error where the linking script cannot find the target under the openocd target

roman65536, 2024-10-10:
First I had the same problem... However it works with stlink-dap.cfg interface file.

Toubidou, 2025-12-03:
Hello，do you solve the problem?please tell me how to solve it