STMicroelectronics/OpenOCD #36: No support for stm32mp2x

WilliamBonilla62, 2025-03-14:
Here's a structured issue report you can submit on the **[[OpenOCD GitHub repository](https://github.com/STMicroelectronics/OpenOCD/issues)](https://github.com/STMicroelectronics/OpenOCD/issues)**.

---

**Title:** Missing `stm32mp2x.cfg` File in OpenOCD - Unable to Flash STM32MP2 Board  

**Description:**  
I am trying to flash my **STM32MP2** board using OpenOCD, but I am encountering an error because the file `target/stm32mp2x.cfg` is missing. This prevents OpenOCD from recognizing the target and initializing the debug session.

---

### **Error Message**
```
Open On-Chip Debugger 0.12.0+dev-00608-gd8ed48fef (2025-02-06-11:20) [https://github.com/STMicroelectronics/OpenOCD]
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
STM32CubeIDE/workspace_1.18.0/AttemptD/CM33/AttemptD_CM33 Debug.cfg:43: Error: Can't find target/stm32mp2x.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 28
at file "/Users/william/STM32CubeIDE/workspace_1.18.0/AttemptD/CM33/AttemptD_CM33 Debug.cfg", line 43
```

---

### **Steps to Reproduce**
1. Install OpenOCD from the official repository or through Homebrew (`brew install openocd`).
2. Create a debug configuration in **STM32CubeIDE** for an **STM32MP2** board.
3. Attempt to flash the firmware using OpenOCD.
4. OpenOCD fails to find `target/stm32mp2x.cfg` and exits with an error.

---

### **Expected Behavior**
OpenOCD should include the necessary target configuration file `stm32mp2x.cfg` for STM32MP2 boards, allowing successful flashing and debugging.

---

### **Actual Behavior**
The required configuration file is missing, preventing the debugger from recognizing the board.

---

### **System Information**
- **OS**: macOS Sonoma 14.x  
- **OpenOCD Version**: `0.12.0+dev-00608-gd8ed48fef (2025-02-06-11:20)`  
- **Board**: STM32MP2  
- **Installation Method**: Homebrew  

---

### **Possible Fix**
- Add the missing `stm32mp2x.cfg` file to the `target/` directory in OpenOCD.  
- If the configuration file is not yet available, provide guidance on an alternative method to flash STM32MP2 boards.

---

### **Additional Context**
I checked my OpenOCD installation folder, and the `target/stm32mp2x.cfg` file is completely missing. Searching through the OpenOCD repository, I also did not find an available configuration file for STM32MP2. Could you confirm whether support for STM32MP2 has been added or if there's an alternative configuration that can be used?

Let me know if you need further information. Thanks! 

---

5. Click **Submit**.

Let me know if you need modifications before submitting! 🚀

ultimo-hombre, 2025-03-14:
I am running into this same issue trying to debug the STM32MP257FAK in CubeIDE v1.18. The error I get is  `Error: Can't find target/stm32mp2x.cfg in procedure 'script' `.

I checked in `/st/stm32cubeide_1.18.0_4/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.3.100.202501240831/resources/openocd/st_scripts/target` and `stm32mp2x.cfg` doesn't exist.

WilliamBonilla62, 2025-03-16:
@ultimo-hombre I would expect them to give support for a devkit that is already out. 

Let's hope the config file is out soon.

rnicolas, 2025-04-11:
In fact, several microcontrollers are not supported by OpenOCD, yours is one, but the STM32C0 family are missing too.

tarek-bochkati, 2025-08-22:
Hello

STM32MP2 config file should be available here
https://github.com/openocd-org/openocd/tree/master/tcl/target/st